mainsail-crew/mainsail #607: Cross control Z axis is reversed

rlaferla, 2022-02-04:
**Describe the bug:**\
The Cross control Z axis is reversed.  

**Expected behavior:**\
Up arrow should raise bed and down arrow should lower bed.

**Screenshots:**\
[image]

**Logfiles:**\
n/a

rlaferla, 2022-02-04:
or if there is a reason for this (I can't think of one readily), make it configurable.

th33xitus, 2022-02-04:
Those are not "Bed-Controls"... those are "Toolhead-Controls".

Arrow down brings the toolhead/nozzle-tip down inside of the printers coordinate system.
Arrow up brings the toolhead/nozzle-tip up inside of that coordinate system. And it doesn't matter if the bed is stationary or can move in z-direction. 

This is no bug and currently the intended behaviour.

rlaferla, 2022-02-04:
Understood but on printers like the Voron Trident where the tool head is stationary (in the Z axis) but the bed moves in the Z axis, it is counterintuitive.  I'd like to see an option to toggle it.

closed 2022-02-05 by rlaferla

meteyou, 2022-02-05:
You can change the arrow direction in your interface settings. But the default direction is like the description from @th33xitus .

th33xitus, 2022-02-05:
@rlaferla For some reason i thought, that inverting the controls is not possible yet for the cross controls. But as @meteyou said, you can already invert pretty much all axis in the settings. Im sorry for this kind of "misinformation".

You have to look for those toggle switches in the Control Settings. You can then invert any axis you like.
[image]

rlaferla, 2022-02-09:
I hadn't seen the Invert Z setting.  I also understand that the control is intended for the direction of the nozzle and not the bed.  I guess it's really a matter of perspective/preference and I'm happy to see that the user is given options.