mainsail-crew/mainsail #608: Webcam feed cuts out when print is started

alexhocks, 2022-02-04:
**Describe the bug:**\
Usually, the webcam feed works fine. As soon as I start a print (i.e. right click a G-CODE file and start the print), the feed disappears. To get it back, I have to reboot the Raspberry Pi. It's gone everywhere, in the dashboard and the dedicated tab as well.

**Expected behavior:**\
The webcam feed is still available when I start a print.

**Screenshots:**\
[image]

**Logfiles:**\
[moonraker.log](https://github.com/mainsail-crew/mainsail/files/8006131/moonraker.log)

**Additional information:**
- Webcam: https://www.welectron.com/Waveshare-15301-OV5648-5MP-USB-Camera-A

**Software Versions:**
- klipper: v0.10.0-250-g01431991
- moonraker: v0.7.1-348-g6749d38
- mainsail: v2.1.1
Initially installed via as Mainsail OS. All system packages are up to date as well.

alexhocks, 2022-02-05:
side question: Why does Mainsail warn me that the print might fail when I restart webcamd while a print is running? Is webcamd in anyway relevant to the printing operation?

lixxbox, 2022-02-08:
regarding your side question:
Afaik depending on the hardware it *could* lead to a cpu spike when the webcam daemon is loading. which *could* stop the print.

meteyou, 2022-02-08:
which webcam mode do you use?

alexhocks, 2022-02-12:
I'm using the MJPEG-Streamer (non adaptive), but right now I cannot reproduce the error. webcamd seems stable. I will monitor if this issue arises again and see whether I can provide more details under what circumstances it happens. Otherwise I'll close the issue.

alexhocks, 2022-02-13:
Okay, I just started a print and the following happened:
Webcam feed was okay. Printer homed and started QGL. At some point during the QGL right after the probe triggered, the print head stops moving. That's when the webcam feed dies. QGL then continues normally and I can get my webcam feed back during printing by restarting webcamd.

meteyou, 2022-02-13:
pls upload your logs!

is it possible, that you have a damaged hardware or a short circuit when the probe triggers and it pulls the 5V line to 0?

but i think this have nothing todo with mainsail itself.. pls ask on the discord for help.

closed 2022-02-13 by meteyou