mainsail-crew/mainsail #609: [FR] Disable History Scrolling in Multiline Command input

rufinus, 2022-02-05:
**Is your feature request related to a problem? Please describe.**
Since #246 was implemented (which is a very nice and needed feature) the history jumping (sorry dont found how it was called exactly) with Cursor+Up & Cursor+Down is not deactivated - so if you want to fix a typo and go cursor+Up your multiline command is replace by the history.

**Describe the solution you'd like**
The easiest fix would be to deactivate the history jumping (or in other words the keypress events) as soon as the input field is changed to multiline.

meteyou, 2022-02-05:
the field is actually always a multiline field. it is only automatically enlarged by the browser. so it is not really clear whether it is a multiline field or not. but since the linux console behaves in exactly the same way, i would keep the current behaviour.

rufinus, 2022-02-05:
> but since the linux console behaves in exactly the same way, i would keep the current behaviour.

that is not the behaviour of the linux console...
 - you cant "enter" linebreaks in the linux console (there is no SHIFT or CTRL magic)
 - if you paste multiple lines in the console EACH line immideattly gets executed and goes into history
 
 > the field is actually always a multiline field. it is only automatically enlarged by the browser.
 Ok the field is always a multiline field - but you still have the `rows` attribute and the style `height` of the element.
 and its not the browser. it is vuetify with most likly some simple CSS. nowadays no JS is needed for this.

meteyou, 2022-02-05:
ok. i mixed it up with something like this:

[image]

there is no ENTER in it... but it should be possible to search for an ENTER in it and stop the history select function.

meteyou, 2022-02-05:
and this would be an multiline input in a linux bash (in my case its zsh)

[image]

rufinus, 2022-02-05:
yes you are right - i forgot the `\` notation. :( *shame on me*

nikolaps17, 2022-10-14:
Hello, Can you assign this issue to me?

PR #2108 (2025-01-03): feat(Console): Improve key up/down function in Console textarea
## Description

This PR improves the key up/down function in the console history to scroll/jump multiline inputs up/down instead of jumping for/back in the history.

## Related Tickets & Documents

fixes #609 

## Mobile & Desktop Screenshots/Recordings

https://github.com/user-attachments/assets/9a630067-142c-42c5-8a00-d0d8c463b2b3

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2025-01-03 by meteyou