mainsail-crew/mainsail #615: [FR] Dropdown with sections in the editor

OasisOfChaos, 2022-02-08:
**Describe the solution you'd like**
It would be nice to have a dropdown list in the editor when editing .cfg/.conf files that shows all the sections in the current file. Selecting an item in the dropdown jumps directly to that particular section. This saves a lot of scrolling (up and) down when you are tweaking your config :-)

TxBillbr, 2023-02-08:
Splitting configurations has a challenge in that any part of the configuration that requires save_config to write to the bottom of the printer.cfg fails if that value is any other file than printer.cfg. This forces a mixture of values in the printer.cfg that really belong in a specialized file such as z_offset in the bltouch.cfg file. I too would find it very nice to have a mechanism to jump to a specific section of a file.

meteyou, 2026-09-25:
This FR is "deprecated". This function already exists a longer time with the sidebar in the editor. This was PR #1943

closed 2026-09-25 by meteyou