mainsail-crew/mainsail #622: [FR] Make TESTZ clickable in output for Z_ENDSTOP_CALIBRATE

rlaferla, 2022-02-09:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
In other console commands, you can click the command in the output to have it inserted as a new command.  Please do this for TESTZ.

**Describe alternatives you've considered**
n/a

**Additional context**
[image]

lixxbox, 2022-02-09:
at a certain point we disabled parsing of console entries to increase performance, as it was very hungry. i don't know if this has changed.

rlaferla, 2022-03-01:
It would be better to have an interactive UI for this anyways.

meteyou, 2022-09-11:
A UI will be added for this feature #1071

closed 2022-09-11 by meteyou