mainsail-crew/mainsail #626: Console does not filter out timelapse take frame messages when timelapse is disabled

MuddyPuddles4325, 2022-02-12:
**Describe the bug:**\
When the timelapse plugin is installed and the console is configured to filter out timelapse messages it does not seem to filter out the "Take Frame" message when the plugin is disabled

**Expected behavior:**\
"Timelapse: disabled, take frame ignored" message would not be shown in the console when the timelapse plugin is disabled

**Screenshots:**\
[image]

**Version:**\
Mainsail v2.1.1

zellneralex, 2022-02-12:
There is a verbose switch in the settings. Setting it to false will omit these messages.

There are 2 types of messages. Mainsail filters setup messages with the Console filter. For macro outputs there is the verbose setting in the setting. 

The reason for that is simple we are sure that there is nearly no need to get the setup messages shown, but you might be interested in the macro output while debug.

th33xitus, 2022-05-15:
Closed due to inactivity.

closed 2022-05-15 by th33xitus