mainsail-crew/mainsail #636: gcode viewer bug

XEL-Maker, 2022-02-14:
**Describe the bug:**\
The results of the g-code preview and the results of the preview in the slicer display different screens.

**Expected behavior:**\
I applied sequential printing options to print two different objects and proceeded with slicing, and as a result of checking them through preview, the g-code viewer showed me like simultaneous printing.

**Screenshots:**\
[image]
[image]

**Logfiles:**\
I added my g-code files
[sequential test.zip](https://github.com/mainsail-crew/mainsail/files/8060610/sequential.test.zip)

Sindarius, 2022-02-14:
The bar on the right of the G-Code Viewer is simply a z height slider not a time slider.  When the file is simulated (coming soon to Mainsail) the file renders sequentially. You can validate the functionality at http://sindarius.com   this is what will be implemented in the near future. 

From your file you can see that the first part is complete and the second one is printing.

[image]

XEL-Maker, 2022-02-14:
I thought it was time, but I first knew it wasn't.
Thank you.

closed 2022-02-14 by XEL-Maker