mainsail-crew/mainsail #643: Bed meshes with spaces or other special characters not working

Misterke, 2022-02-15:
**Describe the bug:**\
The bed mesh / Heightmap view of Mainsail allows renaming the active profile and loading existing profiles from the configuration, but as soon as the profile contains a space (which is possible to provide as in BED_MESH_PROFILE SAVE="PROFILE WITH SPACES"), these actions fail due to Mainsail not quoting the parameters of BED_MESH_PROFILE.

**Expected behavior:**\
Profile manipulation should also work for profiles with spaces in the name.

Misterke, 2022-02-15:
Note also that once https://github.com/Klipper3d/klipper/pull/5222 would get merged, Mainsail could check the actual profiles as maintained in bed_mesh.py rather than those from the static config which don't reflect any changes that have not yet been saved using SAVE_CONFIG.

meteyou, 2022-02-28:
fixed with this commit https://github.com/mainsail-crew/mainsail/commit/96b212e8dc86f58fef6edd02ff10fe7fe36da573

closed 2022-02-28 by meteyou