mainsail-crew/mainsail #644: MJPEG-Streamer (non-adaptive) not working in Safari on macOS/iOS

JRHeaton, 2022-02-15:
**Describe the bug:**\
When selecting the "MJPEG-Streamer" service option for my webcam, it shows a blank webcam view showing "0 FPS" in Safari on macOS and iOS, while it works on Firefox. The experimental adaptive streamer _does_ work in Safari.

**Expected behavior:**\
I would expect that the non-adaptive, "stable" streamer service would be functional in all browsers. This seems to be a regression from past versions of Mainsail.

**Screenshots:**\
[image]
[image]

**Logfiles:**\
[klippy-3.log](https://github.com/mainsail-crew/mainsail/files/8075450/klippy-3.log)
[moonraker-3.log](https://github.com/mainsail-crew/mainsail/files/8075452/moonraker-3.log)

JRHeaton, 2022-03-03:
Any acknowledgment of this issue from the dev team? Seems like a pretty glaring issue for webcam users.

JuicedEagle, 2022-05-08:
I can relate to that. 

mjpeg-stream is not working from iOS on iPhone, wether using Safari, Firefox, Chrome or Edge. 

on Fluidd mjpeg-stream is working without any problems, as well on Octoprint it worked with the normal mjpeg-stream.

meteyou, 2022-05-09:
only because it has the same name, it doesn't mean, that the background is the same. I will not support this mode for ios/safari browsers, because it don't support all JS functions for this mode and other modes also work on ios.

meteyou, 2022-05-12:
today I had time to read up on the problem again and tried various fixes, but unfortunately without success.

it looks like it's a bug in safari itself and it needs to be fixed first: https://bugs.webkit.org/show_bug.cgi?id=215771

jalanjarosz, 2022-05-13:
I'm using "Adaptive MJPEG Streamer (experimental)" on iOS without issues:
[image]

You can also try "UV4L-MJPEG". This shows and image, but I have not tried it. 
[image]

Screen shots are from v2.2-beta1, but I have been using on v2.1.2 as well. 

Cheers
-James

meteyou, 2022-05-14:
In v2.2.0-beta2, we fixed some issues in UV4L mode. This should work also fine now.

meteyou, 2023-06-30:
I'm closing this issue, because safari doesn't seem to fix this problem. furthermore there are enough other streamers that also run with safari and we want to expand the WebRTC support and leave MJPEG behind us.

closed 2023-06-30 by meteyou