mainsail-crew/mainsail #653: Distance Selector in Controls Panel can be Visually Deselected While Still Being Functionally Selected

jneuls, 2022-02-16:
In the Controls Panel, the distance selector button group's active selection, if clicked, will visually deselect while, under the hood, that distance is still selected and, if any of the movement arrow buttons are clicked, a G91+G1 move will still be sent to the connected printer.

[image]
Normal, expected behavior.  A distance is selected, one of the movement buttons is clicked, a G91+G1 move is sent to the printer.

[image]
If you click on the selected distance again (10, in this example) the button deselects.  Clicking one of the movement buttons now, I would expect it to issue an error and send nothing to the printer.  Instead, it just sends a new G91+G1 move to the printer, even though the distance selected is not visible in the user interface.

Interestingly, the Feed amount and Feedrate button groups do not share this behavior.  When an active selection is clicked nothing happens at all.

Digging into the CSS...
When visually selected, the appropriate `<button>` element in the distance selector group gains the `v-item--active` and `v-btn--active` CSS classes, when visually deselected, it loses both.  Over in the Feed Rate group, upon selection, a `<button>` will gain `v-item--active`, `v-btn--active`, and `v-btn--active`.  (That isn't a typo...it literally gains two instances of `v-btn-active`.)  If you click the already selected button, it will lose `v-item--active` and one `v-btn--active`, but it keeps the other `v-btn--active`.

Sounds like there's a bug somewhere in the JS...right where bugs belong.  lol

th33xitus, 2022-02-16:
I was able to reproduce this issue. Thank you for reporting!

jneuls, 2022-02-16:
Quite welcome.  Glad to be of service!  :)

th33xitus, 2022-03-13:
The bug was found and fixed. Will be shipped with the next release!

closed 2022-03-13 by th33xitus