mainsail-crew/mainsail #658: [FR] Clear console.

rlaferla, 2022-02-18:
**Is your feature request related to a problem? Please describe.**
There are at times, too many console entries but I don't want to restart apps.  For example, I may be doing resonance testing and I just want to clear past entries so I can focus on the output from the ADXL.  There are many other cases.

**Describe the solution you'd like**
A button and/or command to clear the console.

**Describe alternatives you've considered**
n/a

**Additional context**
n/a

th33xitus, 2022-02-19:
That is actually something i considered usefull just a few weeks ago. It would be best if this would be integrated into Moonraker first. If we just empty the history in Mainsail, a refresh of the page would bring the entries back, because the console history is served my Moonraker. So Moonraker needs a function like this in the first place.

th33xitus, 2022-02-22:
https://github.com/mainsail-crew/mainsail/pull/672

commit 87b0a8c7 (2022-02-22): feat(console): add the ability to clear the console (#672)

fixes #658
closed 2022-02-22 by meteyou