mainsail-crew/mainsail #666: Preheat option not showing in first time on phone

ruredi87, 2022-02-20:
Hi, firstly I have to thank you for all people who are developing this project. I have problem in gcode screen on phone. I am always using preheat option before start to printing in g code menu, but most of the time it does not showing, generally when I first start browser or after done some printing. I had to reshorting gcode files for see the preheat option also I tired different browsers but it is same.  I recorded a screen capture so you can easier understand. Also English not my native language sorry for mistakes.

**Expected behavior:**\
Show preheat option al the time

[klippy.log](https://github.com/mainsail-crew/mainsail/files/8104737/klippy.log)
[moonraker.log](https://github.com/mainsail-crew/mainsail/files/8104738/moonraker.log)

https://user-images.githubusercontent.com/79092546/154859523-7f343417-4681-4222-905e-908d1d145c0c.mp4

th33xitus, 2022-02-20:
Thank you for reporting. I was able to reproduce this behavior and i consider it a bug. The list needs some kind of re-render so that the preheat option in the context-menu shows up. It does not show up the first time when opening Mainsail.

ruredi87, 2022-02-21:
@th33xitus Thank you so much

closed 2022-04-25 by meteyou