mainsail-crew/mainsail #673: Host Load graph appears to be off by 10x

jalanjarosz, 2022-02-23:
**Describe the bug:**\
Machine Settings Tab, under System Loads, the HOST controller pi graph seems to be off by a 10x factor.   

Assuming that "Load" value is 1=100% and any fraction is an equal percentage (like MCU Load), then the pi graph is off.

In the attached image, my SBC Load is about 15%, when I SSH to the SBC, I'm presented with 11% (it's grabs values at login only), the pi graph should be 15, not 2, as in the image.

**Expected behavior:**\
Pi graph should be 15 to match load value of 0.15, not 2

**Screenshots:**\
[image]

**Logfiles:**\
- klippy.log
[klippy(1).log](https://github.com/mainsail-crew/mainsail/files/8124495/klippy.1.log)

- moonraker.log
[moonraker.log](https://github.com/mainsail-crew/mainsail/files/8124497/moonraker.log)

pataar, 2022-02-23:
It's correct. Just try to run `htop`, `top` or `uptime`. They will all show the load in a 0..1 way.

Example:
[image]
 
Mainsail basically rounds it and multiplies it to get a percentage.

meteyou, 2022-02-23:
and it depends on your cpu. if you have a 4 core cpu, load 4.0 => 100%

jalanjarosz, 2022-02-23:
Yes, the load values match between the text in Mainsail and htop, both are 17%, that is if 1.00 = 100%.  However the pi chart in Mainsail only shows 2%

[image]

pataar, 2022-02-23:
Given you have 8 cores:
`0.17 / 8 = 0.02` which translates to 2% :)

jalanjarosz, 2022-02-23:
Well, That makes sense.   It just was not what I expected.  :-)

jalanjarosz, 2022-02-23:
Closing issue since an answer has been provided.

closed 2022-02-23 by jalanjarosz