mainsail-crew/mainsail #699: Webcam not working

asdfire1, 2022-03-05:
### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

I get no image from camera in mainsail, if I go to the camera address (http://192.168.87.169:8080/?action=stream) I get an image.
Here are my camera settings, I tried putting http://192.168.87.169:8080/?action=stream as url stream and also changing the regular mjpeg to adaptive mjpeg and I get no image from the camera. I'm adding a screenshot of the camera config as well as webcam.txt
[image]
[webcam.txt](https://github.com/mainsail-crew/mainsail/files/8189459/webcam.txt)

### What did you expect to happen instead?

I expected there to be image in mainsail

### How to reproduce this bug?

I installed mainsail os with raspberry pi imager, its a fresh install. I have not used mainsail before.

asdfire1, 2022-03-05:
Same config for both adaptive and regular mjpeg works in Firefox.
[image]

meteyou, 2022-03-05:
do you use also my.mainsail.xyz with chrome? you have to change some flags in chrome if you want to use my.mainsail.xyz.

closed 2022-03-05 by meteyou