mainsail-crew/mainsail #732: Mesh summary/stats data incorrect

dewi-ny-je, 2022-03-23:
### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

"Max" (maybe also "Min") and variance are wrong when a mesh is offset from the origin.

In the example below, the Max should be -0.113, but it's indicated as 0.

[image]

Mesh Leveling Probed Z positions:
-0.142500 -0.112813 -0.140625 -0.180938 -0.132188
-0.178750 -0.137500 -0.180625 -0.196563 -0.165313
-0.210625 -0.169375 -0.197813 -0.218750 -0.149375
-0.216563 -0.148750 -0.182188 -0.187500 -0.119063
-0.261875 -0.211563 -0.203750 -0.196875 -0.121563

### What did you expect to happen instead?

Max should be the actual max value

### How to reproduce this bug?

Any mesh where max is less than 0.

dewi-ny-je, 2022-03-24:
Actually this may be related to the fact that the "scale" selector does not scale the colors between the actually used values, but forces one extreme of the scale to be 0 (which makes sense only when "scale" is not selected).

meteyou, 2022-03-26:
I think I fixed this already in the develop branch. I have to check it.

dewi-ny-je, 2022-05-04:
> I think I fixed this already in the develop branch. I have to check it.

The last release is from February. Is a new one planned soon with the correction?

meteyou, 2022-05-04:
Yes. The first beta will be released soon. We have only to fix some minor issues before we can release it.

meteyou, 2022-05-15:
issue is fixed and beta is published

closed 2022-05-15 by meteyou