mainsail-crew/mainsail #733: Support http page as camera source

adrienlemasle, 2022-03-25:
### Requested feature:

It would be good to add support for any http page as a camera source.

### Solves the following problem:

This would enable users to choose any source for the camera, for example an H264 feed generated with h264rcam (https://gitlab.com/dragonn/h264rcam). The mjpeg option is very bandwith intensive compared to H264.

meteyou, 2022-03-26:
Thx for the FR!

An http source is not the best option in the case of responsive. There is no possibility to resize its content to fit in the current space.

We are generally working on a webrtc player with h264 video streaming (https://github.com/mainsail-crew/crowsnest).

adrienlemasle, 2022-03-26:
Thanks for the reply! I looked at crowsnest but could not see a way to use H264 in the readme. Is this already supported?

KwadFan, 2022-04-10:
Hey there, this looks promising. So, some questions from my side.
Is this h264rcam streamer able to use hardware accelerated encoding of regular usb cams (v4l2) ? 

Could you tryout the branch feat/webrtc of ceowsnest to make it compatible? 

I am highly interestet in that. I would apriciate it if you join our discord ro get in contact with me. Maybe we can work on a solution

closed 2022-04-20 by meteyou