mainsail-crew/mainsail #737: The possiblility to "Drag and drop", straight to the dashboard

DragonFlyer007, 2022-03-27:
### Requested feature:

The possibility to drag and drop gcode files straight to anywhere on the dashboard. (...or spesific area)
Like without having to go to the gcode tab.

(This is possible on Fluidd if i remember correct?)
Really neat function.

meteyou, 2022-05-09:
Fixed it with this PR #777

closed 2022-05-09 by meteyou

DragonFlyer007, 2022-05-23:
Latest Klipper update has now eabled this. Very nice!

th33xitus, 2022-05-23:
> Latest ~~Klipper~~ Mainsail update has now eabled this. Very nice!

You're welcome 😉

DragonFlyer007, 2022-05-23:
> > Latest ~Klipper~ Mainsail update has now eabled this. Very nice!
> 
> You're welcome 😉

😂
Omg, yeah.... Mainsail! 🙈🤦‍♂️

th33xitus, 2022-06-13:
The drag & drop feature of Mainsail has nothing to do with the Klipper installation.

DragonFlyer007, 2022-06-13:
> The drag & drop feature of Mainsail has nothing to do with the Klipper installation.

Yeah, that's what so wierd :? Pure Mainsail, right? Why isn't here anymore then. Hmm