mainsail-crew/mainsail #747: cross movement controls remain locked

shiftingtech, 2022-04-09:
### Browser:

Firefox

### Device:

Mobile device (Phone, Tablet or similar)

### Operating System:

Android

### What happened?

When using cross-style movement controls, the buttons do not enable after homing.

### What did you expect to happen instead?

After homing, movement controls should become available for use.

### How to reproduce this bug?

Home one or more axes of the printer.  Attempt to use movement controls.

### Additional information:

Bug appears limited to the "cross" style controls.  Bar &circle options appear to function normally.

th33xitus, 2022-04-09:
Hi, thanks for the report. It is a known bug and should be already fixed in the current development branch.
So it will be resolved with the next release.

The issue is very likely that you haven't toggled a step size preset below the movement cross. 
So as a workaround for now: Tap one of the step sizes and controls should unlock. There is also a visual bug here, tapping the same preset twice will make the button appear to be not toggled anymore. (that will be also fixed with the next release)

shiftingtech, 2022-04-09:
*sigh*
Yep, that was exactly it.

closed 2022-04-24 by meteyou