mainsail-crew/mainsail #751: Intercept display encoder events

drelich, 2022-04-13:
### Requested feature:

Hi, not sure if it comes here or on Klippet GitHub but I would like an option to intercept display encoder events in a macro.

My specific goal at the moment is to run a macro when I click or rotate the encoder on the display to increase brightness of the backlight neopixel LEDs. I am able to do it as a button on Mainsail dashboard - can dim the LEDs, can restore them to their original brightness, both with a click on that Mainsail dashboard button. But I'd like to use a delayed_gcode that would dim those LEDs after 5 seconds and another gcode macro that would restore brightness as soon as I'd use the encoder... Kinda like the auto-brightness feature on Prusa's MK3 machines where it dims the display after a few seconds of not using the display encoder.

Maybe there's already a way to do it? Did a lot of googling but couldn't find anything. Thanks.

th33xitus, 2022-04-13:
Thats nothing related to Mainsail. And for Klipper support questions, please use either the official Klipper Discord or ask on the Mainsail Discord in the Klipper channel.

closed 2022-04-13 by th33xitus

drelich, 2022-04-13:
Thanks for the pointers.