mainsail-crew/mainsail #752: Webcam site display no configuration available message

Trackhe, 2022-04-16:
### Requested feature:

When users (including me) try to add their webcam, expect that the cam is automatically shown on the webcam site.
But even if the users get the [crowsnest](https://github.com/mainsail-crew/crowsnest) or the old webcam service running, is the site webcam empty. Only the gray box is shown. So sometimes is it not obviously that the webcam needs also configured under the settings.

So Things would be good.
- A message where you write in the empty box, "Add your webcam under the Interface Settings -> Webcams." That `div` is anyway replaced by the cameras if one is configured, so you don't need to write a complex logic for that. (hotfix possible!?)

- A add button that open the Right Webcams settings menu. (Maybe obsolet but maybe also a good option)

- Automatic add default settings when crowsnest detect a capable webcam. (maybe with the settings that configured in the webcam.conf) (only added when no configuration exist) + a settings button under the webcams on the webcam site not on the dashboard

### Solves the following problem:

confused users. 
Difficult to find documentation (makes documentation slightly obsolet) (the best documentation is the one that is not needed.)
Discord inquiries and waste of time. usability

closed 2022-04-18 by meteyou

barecool, 2022-04-18:
FYI - on a slower connection - the message shows even if a cam is configured, until the cams list load from the store.
just wanted to let you know, not a deal breaker

meteyou, 2022-04-18:
Atm, there is no better workflow for that. Because at this time, there is no webcam in the client config. But we plan to hide all elements until mainsail is fully initialised. This feature should fix this issue.

PR #754 (2022-04-18): fix: add error message in webcam panel, if no webcam is available
This PR fix #752