mainsail-crew/mainsail #755: Parts cooling fan override option

drelich, 2022-04-17:
### Requested feature:

Sometimes, especially when I change the speed mid-print, I also need to adjust parts cooling fan speed. So I use the slider, set the new speed and walk away. The issue is that some slicers trigger the fan speed back after, say, bridges which have their own speed set, etc.

In Repetier Server, there is an option to 'force' or 'override' that fan speed for the duration of the print (in fact, it holds that speed until you manually disable it so the fan keeps spinning even after the print is done).

Not entirely sure how RS does it but my wild guess would be it scan active gcode file and replaces any fan speed commands with its own when its pushing the code into the buffer?

Can we have a feature like that in Mainsail, too? Thanks.

meteyou, 2022-04-17:
this is not possible, because klipper have no option to "lock" the fan speed. pls create a FR for klipper. if klipper have this option, we can implement it in mainsail.

drelich, 2022-04-17:
Thanks. I don't think Repetier Server actually relies on firmware to handle that feature. From what I can see not even Marlin has a fan speed "lock". Parsing active gcode from the moment of the "lock" and updating all M106 commands with the required "locked" fan speed, either on the fly or in one go, is perhaps something Mainsail could handle?

th33xitus, 2022-04-17:
> Parsing active gcode from the moment of the "lock" and updating all M106 commands with the required "locked" fan speed, either on the fly or in one go, is perhaps something Mainsail could handle?

No, because neither Moonraker nor Mainsail stream the G-Code and therefore work completely different than OctoPrint or RepetierServer, which DO stream the G-Code. Which enables them to intercept specific commands and or inject their own.

drelich, 2022-04-17:
Right but "something" on the Raspberry Pi is processing gcode that is then "converted" into simple instructions and sent over to the control board to move steppers or open mosfets for heating elements, etc. Is it Klipper itself? Sorry for stupid Qs. With Marlin it's clear what does what. Klipper is pretty new to me (not counting a brief use a few years back). Thanks.

th33xitus, 2022-04-17:
> Right but "something" on the Raspberry Pi is processing gcode that is then "converted" into simple instructions and sent over to the control board to move steppers or open mosfets for heating elements, etc. Is it Klipper itself?

That is Klipper, correct. It's the host part of the Klipper firmware which sends all instructions to the mcu part of the Klipper firmware flashed onto your controller board. Mainsail on the other hand communicates with Moonraker, the Klipper API, and has no **direct** connection to Klipper, nor the ability to read, intercept or inject own G-Code at specific situations.

drelich, 2022-04-17:
Ah, makes sense to me at last lol

Thanks for the explanation! I'll try to suggest that feature to Klipper team.

closed 2022-04-17 by drelich