mainsail-crew/mainsail #756: Multiple Printers Multiplying

rhys100, 2022-04-20:
### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

I have added my 5 printers under printers so I can see an overview.

They keep multiplying and duplicating themselves.

If I delete the duplicates, they seem to reappear.

Picture - https://i.imgur.com/PWd04GV.png

### What did you expect to happen instead?

Not duplicate

### How to reproduce this bug?

.

### Additional information:

klipper
v0.10.0-320-g7e654aed

moonraker
v0.7.1-470-g88b0282

mainsail
v2.1.2

meteyou, 2022-04-20:
Pls explain how to reproduce this issue. i'm not able to reproduce it.

meteyou, 2022-05-09:
closed, because no response from the user...