mainsail-crew/mainsail #757: Print Again Button

rhys100, 2022-04-20:
### Requested feature:

I would like to request a print again button in the dashboard.

Octoprint & Fluid have this and makes reprinting the last print very quick.

### Solves the following problem:

Instead of going to g-code files. then right clicking the top file and hitting print again. Its 2 Screens and 2 clicks more than a print again button.

meteyou, 2022-04-20:
There is already a print again button on the status panel, when the print is complete.

closed 2022-04-20 by meteyou

rhys100, 2022-04-20:
Sorry I should clarify. You are indeed correct - https://i.imgur.com/A43F71l.png

However my end gcode saves the current Z probe offset and restarts and thus that option is no longer there. Is it possible to always display the last print even after a restart so reprint is still an option?

I imagine folks may even save and restart their config after a print and still want to print the same file easily :)

meteyou, 2022-04-20:
this is not really a "reprint", because you reset the state of the virtual_sdcard module of klipper.

we are working on a new status panel for the dashboard with job queue and latest new gcodes on the dashboard, but this will also not fix your issue, because it will only display unprinted/new files.

rhys100, 2022-04-20:
@meteyou what if in that same stats panel you also listed the last file printed and you could click reprint to the right of the file name?

meteyou, 2022-04-20:
@rhys100 i'll take a closer look when i get back to the panel. i haven't really thought of such a usecase until now.