mainsail-crew/mainsail #758: Multiple Webcam Windows

rhys100, 2022-04-20:
### Requested feature:

It would be nice if you could view multiple webcams in a large format rather than 2 small side-by-side windows.

At the moment you can only view 1 large webcam at a time, or split them in half.

Webcam 1 - https://i.imgur.com/CWS47hd.png
Webcam 2 - https://i.imgur.com/XbSJHGb.png
Both Webcams - https://i.imgur.com/ATbSYzL.png

### Solves the following problem:

Instead of rewriting css in SSH it could simple add a 2nd box in the GUI if you add a 2nd webcam in the settings of mailsail.simply

That way viewing remotely via VPN on mobile you can see two large pictures of both webcams

rhys100, 2022-04-20:
It could also be cool to pull in extra windows from other printers in dashboard or show in the printers gui where all printers show the gcode preview

meteyou, 2022-04-20:
> It could also be cool to pull in extra windows from other printers in dashboard or show in the printers gui where all printers show the gcode preview

do you really want to kill your PC with 5 times gcode viewer?

meteyou, 2022-04-20:
> Instead of rewriting css in SSH it could simple add a 2nd box in the GUI if you add a 2nd webcam in the settings of mailsail.simply

if you want to change anything via CSS, pls use the custom CSS file (https://docs.mainsail.xyz/features/theming/custom-css) instead of changing anything in the original source code. this will be overwritten with every update!

closed 2022-05-30 by meteyou