mainsail-crew/mainsail #761: Manual Extrude greyed out

grondo14, 2022-04-22:
### Browser:

Firefox

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

Hi all! Just getting a Makergear M2 from 2013 set up with Mainsail and Klipper! So far having a blast!

I’ve run into something that feels like a bug but might be a feature… as they say…

When I get the hotend up to temp for manual extrusion whilst setting up the printer.cfg file, I find a setting that needs to change, like the gear ratio. I go to printer.cfg while the hotend is still hot, make the change, and hit “save & restart”, all is well. The system reboots, comes back up nice and pretty, senses the hotend is still at 200C, and shows it beautifully start to drop off because my command to 200C was wiped with the reboot, as expected.

However, the “extrude” button on the Controls window of the Dashboard is greyed out and says It won’t extrude until the hotend is over 170C (a reasonable limitation). So, knowing that the hotend is now at 194C and dropping, I re-command the hotend to 200C, which it kindly obliges. Only, the “extrude” button is still stubbornly unresponsive and still requests the extruder be above 170C.

It isn’t until the extruder is allowed to cool below 170C and reheated to above 170C that the button comes back on. I smell a state machine issue where the “extrude” button only looks for a transition from 169 to 170C after boot, instead of checking for the current temp on boot.

Am I on to something? or am I missing something simple?

Thanks!

### What did you expect to happen instead?

I expect that if the hotend is above the safe extrude threshold that it will allow extruder to be commanded.

### How to reproduce this bug?

1. turn system on and preheat to 200C or something above safe melt temp
2. restart klipper/mainsail - hotend temp command set back to 0, but hotend still at ~195
3. attempt to extrude plastic (button greyed out even though the hotend is above safe temp)
4. command hotend to maintain 200C.... still no manual extrude allowed
5. turn off hotend, allow to cool below safe extrude temp.
6. reheat hotend above safe temp, the button now allows extrusion.
7. Would be interesting to try to start a print in the failed state.

meteyou, 2022-04-22:
thx for your bug report. mainsail uses the value "can_extrude" from klipper. there is no "temperature matching" in mainsail.

I think it is more a "reactive" issue with vue in mainsail. mailsail have to check which extruder is currently active.

there is also some active work on this panel: https://github.com/mainsail-crew/mainsail/pull/711
we will double-check this issue again with the new extruder panel with your detailed reproduce description.

grondo14, 2022-04-22:
Thanks Stefan.
I'll look forward to seeing the new tool!

meteyou, 2022-04-25:
I double-checked it with the new extruder panel #35, and it looks fixed.

Thx for your bug report!

closed 2022-04-25 by meteyou