mainsail-crew/mainsail #769: hidden Macros don't work

Whitehawk2000, 2022-04-28:
### Browser:

Chrome

### Device:

Laptop

### Operating System:

Windows

### What happened?

I wanted to hide some Macros in the Mainsail UI ,that will never be used, instead of adding them into a group where they still show up.
So i put an underscore in front of the name, like desribed here: https://docs.mainsail.xyz/features/hide-gcode_macros
After that the Macro disappeared in the UI, but it also did not work anymore, when typed into the console.
I think thats not supposed to happen. I also do not know if this is a Mainsail or Klipper related problem.

### What did you expect to happen instead?

As i understand, if you hide a macro it shouldn't be shown in the UI but should still work.

### How to reproduce this bug?

Adding an underscore before the Macro Name.

th33xitus, 2022-04-28:
With prefixing a macro with `_` you basically changed its name. So you also have to call that macro with that exact, new name.
`my_macro` != `_my_macro`

There is another option in Mainsail to hide macros. You find it in the settings in the macros section. There you can either toggle the visibility of any macro in the "simple" management method, or when changing to expert, you can create your own (multiple if you wish) macro panels. Any macro not assigned to a macro panel in expert mode won't be visible on the dashboard.

Whitehawk2000, 2022-04-28:
Thank you for the fast response. Should have thought about that. :-) 
I think this solution was simply to counterintuitiv for me.  I mean, who would think, that you have to correct now every Macro, that calls these Macros, just to hide them. (in my case mostly M300-notification-sounds that are used nearly in every other macro).

I already made groups for the macros. It is a really nice function, but i wish i could backup these Mainsail-related configurations just like downloading the printer.cfg. A mainsail.cfg in the same directory would be nice!

th33xitus, 2022-04-28:
Typically prefixing existing macros with `_` only for hiding them is not the intention behind that. Thats why there are those toggle switches. Prefixing with `_` is usually sometimes done in programming to mark specific functions as internal ones, that won't ever (and should never) get accessed from the outside. Those habits also found their way into writing Klipper macros once everything gets a bit more complicated etc.. That's why there is that "automation" so to speak for automatically hiding underscore-prefixed macros which usually are not intended to get invoked on their own, and rather be invoked by **_another_** macro.

> i wish i could backup these Mainsail-related configurations just like downloading the printer.cfg
Go into Settings -> General -> Moonraker DB -> Backup
You can then tick some checkboxes to backup either specific parts of your Mainsail settings or tick them all and make a full backup of all currently applied settings.

Whitehawk2000, 2022-04-28:
I know why it is done this way... I can deal with it. I am just saying that it is counterintuitive. Especially for those people, that cannot programm and that are newbies to Klipper. It is just a suggestion to make everything even more userfriendly.
As well as the backup. It is really nice to have this option somewhere hidden in the options, but why this fragmentation? All configs should be in the main config folder of klipper, as for now this is the heart of Klipper.
As i said, this is just a suggestion to make everything even more user friendly as it is already.

th33xitus, 2022-04-28:
> Especially for those people, that cannot programm and that are newbies to Klipper

I fail to follow that comparison. The topic was about hiding macros. And the preferred method is by toggling the switches and not by prefixing existing macros with an underscore as it changes the name inside Klippers scope. Im not quite sure i understand what you suggest to make it even more userfriendly in Mainsail as i see an individual toggle button per macro as pretty userfriendly and no programatic way of hiding is necessary -> no programming knowledge needed at all.

> All configs should be in the main config folder of klipper, as for now this is the heart of Klipper.

Correct, it is for Klipper. But Klipper and Mainsail follow completely different strategies of getting configured **and** how those configurations are stored. It probably would be possible to automatically export the backup into Klippers configuration folder, but again... then that file lies on your Raspberry Pi (or generally speaking on your host). That is not a real backup. A real backup is made to external, independent storages. Thats why you can download Mainsails settings to your PC by using that backup function. And you didn't mention what you need that backup for. All settings are stored in the Moonraker database by default anyways. The backup function of Mainsail is usually helpfull if you want to have the same settings applied to Mainsail for a printer on a different host, with its own individual Moonraker database.

Whitehawk2000, 2022-04-28:
Hey, you understood it. :-) Exactly if you want to take the config to another host or if you need to do a fresh install. SD-Cards love to die... The easiest way is to have ONE place, from where you can backup ALL configurations from.
Now you have fragmentation, because you have to go to at least 2 places to backup files. Not good for beginners. By the way the definition of "backup" is irrelevant for that. 
With 25years experience in programming, electronics and other related topics, i can deal with that. But it can be done better.
And remember, that is no offending from my side, just a suggestion which encountered as a side effect from the main topic!!!

th33xitus, 2022-04-28:
> With 25years experience in programming

Looking forward for your proposal :)

Whitehawk2000, 2022-04-28:
Told you that proposal already: ONE place, from where you can backup ALL configurations from. (MainsailUI-config AND Klipper-config Folder).
At the end of the day it is your version of a UI for Klipper, so you decide if you listen to this suggestions or not.

closed 2022-04-28 by Whitehawk2000