mainsail-crew/mainsail #773: Show Z-Offset in print history (from printer cfg and manually adjusted)

tispokes, 2022-05-03:
### Requested feature:

Would be nice to be able to see the absolute Z-Offset and also the Z-offset which was manually set in the dashboard.

### Solves the following problem:

Sometimes I need to adjust the Z Offset on different printbeds, would make it easier.

### Additional information:

Maybe also a nice histogram for Flow and Speed settings. (y)

meteyou, 2022-05-03:
Hey! thx for your request. There are multiple problems with integrating this/these feature(s).

At first, Klipper has no "absolute z-offset". Klipper only works with endstop positions and gcode offset (this is no z-offset).
And the second issue is, that has to be first implemented in moonraker. mainsail just display the history, but moonraker store all data.

tispokes, 2022-05-03:
Okay, then I should open the FR on moonraker? Or is the Ref Label enough?

meteyou, 2022-05-03:
@tispokes pls create an FR in the moonraker repo and add a link to this FR.

closed 2024-12-30 by meteyou