mainsail-crew/mainsail #775: Fan slider that overrides fan G-code for duration of print.

PhilBaz, 2022-05-06:
### Requested feature:

Fan slider currently overrides the value until a new G-code value is received, this can be on the next layer. 

It would be great to have a fan slider/toggle that overrides fan G-code for duration of print.

### Solves the following problem:

If you are adjusting the fan to save a print, this would allow you to save the print and not worry that G-code will reset the fan to a problematic value.

I often use regular/maximum fan speed with threshold values in Cura. This can lead to fan values bouncing around a lot depending on the layer geometry. As such, currently the fan slider doesn't accomplish much for a print in progress as it will be reset quickly by g-code.

### Additional information:

maybe this is a klipper thing more than a MS thing?

meteyou, 2022-05-06:
As you wrote at the end, this is a Klipper thing. Mainsail only sends gcode to Klipper to change things like the fan, but Klipper has no method/function to "lock" it. When your gcode has the following fan command, the fan speed will be changed.

th33xitus, 2022-05-06:
Duplicate of: https://github.com/mainsail-crew/mainsail/issues/755

PhilBaz, 2022-05-06:
@th33xitus Makes sense ill bump this over to the Klipper git...

Thanks!

PhilBaz, 2022-05-06:
@th33xitus  I presented the idea on the Klipper discord server.  Hopefully someone picks it up. I don't know how to code so cant contribute myself, unfortunately. I think its really a worthwhile feature for Klipper.

PhilBaz, 2022-05-08:
@meteyou  Some one at the Klipper discord responded to the idea saying the following (below)...... Perhaps this can be implemented behind the scenes in mainsail so its just an interface toggle?

..................................................
@PhilBaz You can definitely accomplish what you are looking for with a gcode_macro using the rename_existing and use a variable in the macro to set that you have used an override value and it can block (or modify) any gcode fan commands for the rest of the print. A rough example:
[gcode_macro M106]
variable_override: 0
rename_existing M106.1
gcode:
    SET_GCODE_VARIABLE MACRO=M106 VARIABLE=override VALUE=(params.R|default(0)|int)
    {% if (override|int) == 0 %}
        {% if 'S' in params %}
            M106.1 S{params.S}
        {% endif %}
    {% endif %}

You can use this either by:
M106 S(temp_here)
M106 R1

or
M106 S(temp_here) R1
if you ever want to release the override you can do:
M106 R0
So if you use a slider or function in Mainsail (I haven't switched over yet so I don't know the interface) you can add a button that sends M106 R1 to lock in the last thing you set

meteyou, 2022-05-08:
@PhilBaz we only implement Klipper features. something like this is a little bit wired because the name matching and and and...

PhilBaz, 2022-05-11:
@meteyou 

I understand. Native klipper features only.

Im not sure where to propose the idea for Klipper development after looking at their Git. I tried the discord.

Do you think there would be value in this feature for Klipper, or is it just me? If so, maybe you can put the idea in the right place so it will be seen by the right people.

PhilBaz, 2022-05-11:
@meteyou I hope you don't mind if I post a screenshot of your last reply there? Just to let them know MS crew also feels it could be valuable and would likely integrate the feature into the UI if it were developed.

meteyou, 2022-05-11:
@PhilBaz feel free to add a screenshot or better a link to this FR. pls also post a link to the post in the Klipper discourse here.

PhilBaz, 2022-05-11:
@meteyou 

Done! Thank you...

meteyou, 2024-12-30:
I will close this FR, because it needs first a implementation in Klipper itself. If there is something new in Klipper, pls post it here and we will reopen it.

closed 2024-12-30 by meteyou

PhilBaz, 2024-12-30:
@meteyou Thanks. I understand. I brought it up on the klipper discourse and discord. No bites.