mainsail-crew/mainsail #776: In the 'printing' dialogue, the max flow (flyout that appears when hovering on flow) doesn't show anything above 9.9mm/s^3

PhilBaz, 2022-05-06:
### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

In the 'printing' dialogue, the max flow (flyout that appears when hovering on flow) doesn't show anything above 9.9mm/s^3.

This flyout info should contain the max flow attained during the print, but it appears to be caped at 9.9mm.

### What did you expect to happen instead?

It should show the actual max flow achieved. 

Perhaps this should have some minimum time calculation to avoid reporting hypothetical flow spike/artifacts that sometimes show briefly. 

### How to reproduce this bug?

Print over 10mm^3 and check the flyout/hover max flow report.

### Additional information:

Thanks!!!

meteyou, 2022-05-15:
thx for your report! i fixed this issue in this bigger PR: https://github.com/mainsail-crew/mainsail/pull/726

PhilBaz, 2022-06-05:
@meteyou 

Hi, I just tried out the beat. Switched to beta for the first time, just looked to good to wait.

The issue is fixed, but I think the flow rate now needs some form of smoothing or something, to knock out synethic/fake/theoretical spikes.  

Currently I'm getting a max flow rate of 43mm^3, while my first layer is going down. But ill never get anything close to that in the whole print. So whatever is causing that synethic/fake/theoretical spike is essentially making the max flow value useless before the print has even really picked up.

Not sure what the solution is....I imagine some form of min time threshold.

Also, ability to reset the max while printing would be great for testing prints and tuning certain sections. So I can track the max over a certain part. that would be nice.

Thanks again!

meteyou, 2022-06-05:
43 will be your retract. To reset this value, just refresh your browser tab.

closed 2022-06-05 by meteyou

PhilBaz, 2022-06-06:
@meteyou 

Hey, I hear what you are saying about the retract......But, its still ruining the feature. Can retract be excluded?

meteyou, 2022-06-06:
There is no possibility to detect a retract/detract.

PhilBaz, 2022-06-06:
@meteyou  Is a minimum time filter possible, to filter them out?  It would be great to see max flow, but the retract is essentially rendering it useless.

PhilBaz, 2022-06-08:
I was just rethinking this...of course it couldn't just cap the values to a threshold, it would just be the same problem again...it would have to discard values over a threshold.