mainsail-crew/mainsail #778: Print Statistics And Thumbnails Not Showing

Alexander-T-Moss, 2022-05-07:
### Browser:

Safari

### Device:

Laptop

### Operating System:

macOS

### What happened?

This very well be a "me being stupid" situation but I can't seem to find out why print file statistics and thumbnails aren't showing at all. The files are printing fine but they are not doing anything else in regard to outputting additional info it seems.
[image]

### What did you expect to happen instead?

To see the print statistics being filled out and code thumbnail previews instead of blank file icons

### How to reproduce this bug?

I wouldn't know :)

th33xitus, 2022-05-07:
Can you please upload one of the two files we can see on the screenshot? In case those files contain sensible data you can't or don't want to share, can you please provide a different file where this issue also occurs?

Alexander-T-Moss, 2022-05-07:
Here you go, just changed the extension to .txt to upload it to git :)

[main_body-beta1.txt](https://github.com/mainsail-crew/mainsail/files/8645801/main_body-beta1.txt)

Alexander-T-Moss, 2022-05-07:
I sent one of the files over to someone who uploaded it to their printer running mainsail and can confirm it displayed all the stats and thumbnail as expected on their's

th33xitus, 2022-05-07:
I quickly had a look inside the file and it looked completely fine.
I then browsed to http://my.mainsail.xyz and uploaded the file to my development printer... it gets parsed completely fine and i see all metadata:
[image]

So im not able to reproduce the issue on my end.

One suggestion i can make:
Restart the Moonraker service... or the complete host... the metadata should get re-parsed then afaik.

Alexander-T-Moss, 2022-05-07:
Okay, will restart the host once this print finishes - should be about 10 mins - and will report back :)

Alexander-T-Moss, 2022-05-07:
Yeah, restarting the host and turning the printer off/on don't seem to do anything :/

th33xitus, 2022-05-07:
For now i think we can rule out a bug in Mainsail and i will close this issue.
Please visit us on Discord, as there may be other issues not related to Mainsail. Maybe with Moonraker.
https://discord.gg/mainsail

closed 2022-05-07 by th33xitus