mainsail-crew/mainsail #786: it's not a bug. It's a feature. we switched the scrollbars to JS scrollbars to have more space in the gui.

bobbbyqwa, 2022-05-09:
it's not a bug. It's a feature. we switched the scrollbars to JS scrollbars to have more space in the gui.

_Originally posted by @meteyou in https://github.com/mainsail-crew/mainsail/issues/772#issuecomment-1121368806_

bobbbyqwa, 2022-05-09:
Hello I am opening this issue because I believe it to be closed prematurely. I respectfully think there needs to be some due diligence. To better explain what I am talking about I will post a video through a icloud link. It will take a little time to upload

https://www.icloud.com/photos/#013aYIqgHDl-aDRoHuQ9AzV2w

meteyou, 2022-05-11:
thank you very much for your video! now I know what you mean! I would decelerate this as a feature request.

yeah atm, the scrollbar will only be visible, during the scrolling movement and will hide after a short delay. we use this library for this scrollbar: https://github.com/KingSora/OverlayScrollbars

I will check the library if it is possible to "auto display on hover" the scrollbar.

th33xitus, 2022-05-11:
The scrollbar will now appear as soon as the mouse is moved. This should fix your issue. We generally think, that this behavior is a better approach.

PR #793 (2022-05-11): refactor: display scrollbar when mouse is moving
this pr fix #786
closed 2022-05-11 by th33xitus