mainsail-crew/mainsail #788: Kernel panic: Unable to mount root fs on unknown-block(179,7)

witmanto, 2022-05-11:
### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

I haven't been able to boot my Pi for two weeks. Everytime I boot it gets stuck in
end kernel panic - not syncing: VFS: Unable to mount root fs on unknown-blick(179,2)
I tried using a keyboard to enter recovery mode using shift but the keyboard doesn't get recognized.

I tried to edit the file cmdline.txt in the boot fs from my linux desktop and changing the line
root=/dev/mmcblk0p2

### What did you expect to happen instead?

na

### How to reproduce this bug?

na

th33xitus, 2022-05-11:
Not Mainsail related. Invalid issue.

closed 2022-05-11 by th33xitus

Trevo525, 2023-11-08:
Is MainsailOS not maintained directly by the Mainsail team? I cannot get past this issue. I didn't install anything externally. All that I have to do to get here is Install MainsailOS using Raspberry Pi Imager. Run all updates using [Mainsail Update Manager](https://docs.mainsail.xyz/setup/updates/update-manager). Then, if I restart my pi it will boot into this issue. 

[image]

meteyou, 2023-11-08:
@Trevo525 same team, but this is the wrong repo for MainsailOS issues. but in general we do not use the github issue tracker for support, but only to manage the project (bugreports & feature requests). for support please use our discord.

PS: but the error looks to me as if your SD card/filesystem has a defect.

Trevo525, 2023-11-08:
Thanks, I have tried three microSD cards that were all brand new and this happens everytime. I have it working now using [KIAUH](https://github.com/dw-0/kiauh) instead. You can close this now though.