mainsail-crew/mainsail #799: PWM Fan Flyout has dual check boxes for 'Show PWM speed in chart'

TodWulff, 2022-05-13:
### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

https://i.imgur.com/chPCsEa.png

Redundant 'Show PWM speed in chart' control in PWM Fan Flyout from Temperatures panel.

### What did you expect to happen instead?

Have only a single check box for showing/hiding the PWM speed.

### How to reproduce this bug?

Have a PWM Fan configured in Klipper and then in Mainsail, click a PWM Fan Icon or Name in the temp panel.  Flyout depicts dual 'Show PWM speed in chart' checkbox controls

### Additional information:

Noted this on 2.2.0-alpha and mentioned it.

th33xitus, 2022-05-13:
Hi, thanks for reporting that issue!
I think i found the culprit and made a PR with a fix for it.

If you are familiar with manually installing Mainsail and don't mind trying to check out the fix, you can download an automatically generated build from here:
https://github.com/mainsail-crew/mainsail/actions/runs/2319144474

Feel free to install it temporarily for checking if the issue was also resolved for you. Feedback is always appreciated!

TodWulff, 2022-05-15:
I'll endeavor to do so today.  Thanks for the quick turn on a possible resolution.

closed 2022-05-15 by meteyou

TodWulff, 2022-05-16:
> I'll endeavor to do so today. Thanks for the quick turn on a possible resolution.

I confirm that this looks to be fixed in 2.2.0-beta3.  Thank you!

commit 97912022 (2022-05-15): fix: duplicate checkbox for pwm fan (fixes #799) (#802)