mainsail-crew/mainsail #800: Unable to print after update

soleshoe, 2022-05-13:
### Browser:

Chrome

### Device:

Laptop

### Operating System:

macOS

### What happened?

Mainsail OS was working fine until I made an update this morning. The cura2moonraker plugin gets the following error when sending an object to the printer : 

**Moonraker error:**
There was a network error: 401 error transferring http://mainsailos.local/server/files/upload - server replied : internal server error

When I check in mainsailOS UI, the object is present in the "GCode files" section, but if I click on the object and then on the "START PRINT" button, nothing happens.

I tried rebooting printer + raspberry pi and problem still present.

### What did you expect to happen instead?

Printing GCode stored !

### How to reproduce this bug?

100% reproductible with 13/05/22 update around 8am CEST

th33xitus, 2022-05-13:
We are aware of that issue, but this is not a Mainsail issue.
It is due to a recent Moonraker update.

You need to roll back Moonraker by at least two commits. SSH into the host and do the following:
```
cd ~/moonraker
git reset --hard HEAD~2
sudo systemctl restart moonraker
```
Do not update Moonraker afterwards for now.

soleshoe, 2022-05-13:
Perfect, working again, thanks 👍

closed 2022-05-13 by th33xitus

th33xitus, 2022-05-13:
PSA: Arksine pushed a fix for the recent issue. Update Moonraker again and everything should work again.
https://github.com/Arksine/moonraker/commit/0164bb239ea1032e105428435ff2643378bf4bfa