mainsail-crew/mainsail #807: System Loads Text color black on dark grey .

gurusonwheels, 2022-05-15:
### Browser:

Firefox

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

Text in System Loads is  now almost unreadable as color of text is black on a dark grey background . 
[image]

### What did you expect to happen instead?

status to be readable .

### How to reproduce this bug?

load system loads .

th33xitus, 2022-05-15:
Are you using a custom theme?

gurusonwheels, 2022-05-15:
> Are you using a custom theme?

not that im aware of ... how would i know ?

th33xitus, 2022-05-15:
Check your klipper_config folder for a folder named `.theme`. And if it exists, there is probably a css file in there.

gurusonwheels, 2022-05-15:
> Check your klipper_config folder for a folder named `.theme`. And if it exists, there is probably a css file in there.

went in KIAUH and done remove theme ...  works as should . do not remember ever installing a theme . did not even know that existed .   thanks

closed 2022-05-15 by gurusonwheels