mainsail-crew/mainsail #818: {FR] Suppress null RPM on temp chart for Fans w/o tach input defined

TodWulff, 2022-05-17:
### Requested feature:

https://i.imgur.com/xOmwF1T.png

If possible, when a fan doesn't have a tech input defined, can the 'null RPM' be suppressed on the Temp table.

Just a low priority nice to have.  TIA for any consideration.

### Solves the following problem:

Simply provides a more elegant less cluttered implementation.

th33xitus, 2022-05-17:
Thanks for the request. That makes definitely sense to me!

th33xitus, 2022-05-17:
I already wondered why that should even happen and display null. Now looking at the code, it was definitely not intended to be shown! So you found a bug im going to fix! I will re-label this issue from a feature request to a bug report.

Thanks again!

commit 76cedb17 (2022-05-18): fix: do not show `null RPM` in temp chart (fixes #818) (#820)
closed 2022-05-18 by meteyou