mainsail-crew/mainsail #845: [v2.2.0-beta4] impossible to set pressure advance values for extruder_stepper

th33xitus, 2022-05-26:
### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

Currently it is impossible to safely set pressure advance values for printer configurations where multiple extruders are configured with `[extruder_stepper]` sections in the printer.cfg. Using Mainsails input field for pressure advance will fail, as it tries to assign those pressure advance values to an `extruder`, that is not allowed to take pressure advance values. 
[image]
[image]

Even though an `extruder stepper` might be synced with an `extruder`, `extruder1`, `extruder2` and so on, it is currently impossible to figure that out which `extruder stepper` is synced with which `extruder`. That is due to technical limitations of Klipper, as Klipper (+Moonraker) does not offer a safe way to figure out which `extruder stepper` is currently the active one.

`[extruder]` is not allowed to have pressure advance values configured **and** won't accept any pressure advance values send to it via g-code. Klipper in turn requires the pressure advance to be set via g-code **and** by specifying the correct `extruder stepper` name, rather than the `extruder` name it is synced to.

After joint consultation with @pedrolamas (Fluidd developer) on how to tackle that issue we came to the following conclusion:
We will be unable to allow setting pressure advance values via input fields as long as Klipper offers a suitable solution to the problem above. For the meantime it is probably the safest solution to hide those input values once it's clear that there are `[extruder_stepper]` defined in the users printer.cfg. We have to wait for Klipper upstream changes to happen. Only then we can re-enable those input fields again.

### What did you expect to happen instead?

-

### How to reproduce this bug?

-

jalanjarosz, 2022-06-30:
Not sure if this info would help with creating a solution.  Here is a complete config setup of a multi-stepper extruder using extruder_stepper:

https://klipper.discourse.group/t/x-in-1-out-non-mixing-extruder-config/2387

Pressure Advance is set via the following command:

SET_PRESSURE_ADVANCE ADVANCE=0 EXTRUDER=[tool_name]

th33xitus, 2022-06-30:
It's not about that we don't know how to set pressure advance for an extruder_stepper, it's about a technical limitation of Klipper which doesn't allow us to figure out which extruder_stepper ( -> `[tool_name]` ) is currently the active one.

noahliketheark, 2022-10-18:
> It's not about that we don't know how to set pressure advance for an extruder_stepper, it's about a technical limitation of Klipper which doesn't allow us to figure out which extruder_stepper ( -> `[tool_name]` ) is currently the active one.

Is the following command new to the latest Klipper release?
printer[printer.toolhead.extruder].target

th33xitus, 2022-10-18:
> Is the following command new to the latest Klipper release?
printer[printer.toolhead.extruder].target

@noahliketheark 
Can you explain why you chose this issue to ask the question? I can't make a connection between your question and the actual topic right now.

noahliketheark, 2022-10-18:
> > Is the following command new to the latest Klipper release?
> > printer[printer.toolhead.extruder].target
> 
> @noahliketheark Can you explain why you chose this issue to ask the question? I can't make a connection between your question and the actual topic right now.

The problem with including pressure advance in the GUI for a 2-in-1-out setup is that Klipper doesn't extuder_stepper is the active one as I understand it. If printer[printer.toolhead.extruder].target is a new feature, would this allow the pressure advance UI to work?

th33xitus, 2022-10-18:
Ah now i understand what you mean. No this is not a newly implemented status object.
` printer[printer.toolhead.extruder].target` is not even what you think it might be. It would give you the target temperature of the current extruder. So in case of a 2in1 setup, it just references to the thermistor reading of that one hotend you have there.

closed 2025-11-26 by meteyou