mainsail-crew/mainsail #857: Edit temperature on Mobile Device

SchwabJ, 2022-05-28:
### Browser:

Chrome

### Device:

Mobile device (Phone, Tablet or similar)

### Operating System:

Android

### What happened?

Hello, 
with v2.2.0-beta3 i cant edit the temperature (extruder, heatbed,...)on my phone (Pixel 6 Pro, Android 12, Chrome) . After type in a new temp nothing happens.

### What did you expect to happen instead?

Set the new temp like on desk. 

### How to reproduce this bug?

Pixel 6 Pro, Android 12, Chrome

### Additional information:

[image]

th33xitus, 2022-05-28:
Please update to beta4.
That was a bug, and it got fixed with beta4.

meteyou, 2022-05-28:
issue was fixed with this PR: https://github.com/mainsail-crew/mainsail/pull/828

closed 2022-05-28 by meteyou