mainsail-crew/mainsail #869: Is something off with the extruder section calculation?

PhilBaz, 2022-06-06:
### Browser:

Chrome

### Device:

Laptop

### Operating System:

Windows

### What happened?

 My normal extrusion setting of  5-7mm/s is now producing clicking....I have been using these settings a while. 

Also, the new calculation says I should expect 109mm of filament for 25mm requested. Maybe I'm misunderstanding something.

[image]

### What did you expect to happen instead?

 No clicking extruder, as usual....

The calculation should show the requested amount of filament. 25mm requested should show estimated extrusion 25mm @ Xmm^3

### How to reproduce this bug?

extrude?

meteyou, 2022-06-06:
If you use the correct nozzle size on your config, it should be 109mm extruded filament.

closed 2022-06-06 by meteyou

PhilBaz, 2022-06-06:
@th33xitus 

Do you happen to know the calculation?   I'm curious to understand.

PhilBaz, 2022-06-06:
Ahh okay.....its volumetric.      Its listed as mm currently. Perhaps labelling it mm^3 would eliminate some confusion for others.

                     v-if="filamentDiameter && nozzleDiameter"
                        style="font-size: 0.8em"
                        class="text--disabled text-caption font-weight-light d-flex justify-center">
                        {{ $t('Panels.ExtruderControlPanel.EstimatedExtrusion') }} ~ {{ extrudedLength }} mm @
                        {{ volumetricFlow }} mm³/s
                    </div>

get extrudedLength(): number {
        return Math.round(this.feedamount * this.extrudeFactor * (this.filamentDiameter / this.nozzleDiameter))

PhilBaz, 2022-06-06:
No, I'm still missing the point....

Its extruding 25mm of filament. The max volumetric that could be is 60mm^3.

Sorry, I'm just trying to understand....what measurement is the 109mm ?

PhilBaz, 2022-06-06:
@th33xitus

Yes, I thought volume, not volumetric flow rate, but absolute volume.

 Ahhh, its the length coming out the tip of the nozzle. Sorry, have never even thought about it from that angle before.  

Got it...thanks...sorry.

th33xitus, 2022-06-06:
I thought i made it clear with:
> The calculation of the estimated extrusion is to be seen after the nozzle orifice.
25mm of requested filament do not translate to 25mm of extruded (molten) filament.

in this comment https://github.com/mainsail-crew/mainsail/issues/869#issuecomment-1147527344 😄
anways, glad you understand now.

PhilBaz, 2022-06-06:
@th33xitus 

You did, but the switch in my brain hadn't flipped yet....you made it clear, for sure.