mainsail-crew/mainsail #870: Would like to see a option to put g-code viewer on the Dashboard just like webcam

karlalbury, 2022-06-06:
### Requested feature:

Add option to pull the Gcode Viewer Tracking window to the Dashboard.
find my self bouncing between the two a lot.

Include Render of what looking for.
[image]

bnieves1994, 2022-10-05:
I would also love to see this.

piwi3910, 2022-11-02:
I would also like to see this, The feature exists in Fluidd, something similar in mainsail would be great

Sindarius, 2023-01-06:
I recommended against this because the 3D viewer is much more resource intensive compared to a 2d viewer. I think an alternative would be to bring some of the panels like progress and temps into the viewer page.

Nazgile94, 2024-01-12:
> I recommended against this because the 3D viewer is much more resource intensive compared to a 2d viewer. I think an alternative would be to bring some of the panels like progress and temps into the viewer page.

its optional, if you dont like it - dont add the widget to the dashboard

i +1 this, would be nice ! maybe with toggle for 3d/2d

PRGeno, 2024-05-12:
I agree with adding a 2D version of the GCode viewer to the dashboard especially for making Excluding Objects more useful and accessible.  I don't think the 3D viewer makes much sense on the dash, but for me a 2D viewer like Fluidd is a critical feature and one of the only reasons I typically use Fluidd over Mainsail.  

I think having the bed mesh on the dash like Fluidd would be nice to have too, but no where near as critical as a 2D GCode viewer.

Jeppedy, 2024-11-03:
I'd be happy with the OctoPrint-style GCode Viewer.  One layer at a time, showing what will be laid down for that layer, and allow scrubbing to see future/past layers.  A nice simple 2D viewer compared to what Mainsail has right now.

dot-Justin, 2024-12-25:
+1 this, would love to see.

sjoerd104, 2025-02-09:
+1 would be great, even octo's 2d one.