mainsail-crew/mainsail #883: Temperature graph State

numindast, 2022-06-10:
### Requested feature:

The temperature graph currently displays nozzle and bed temperature, and works well.

Can we also have, or access a toggle for, plotting the heater State %'s too?

My reasoning: one can tell at a glance when the PWM power state to the heaters has levelled off. This would be a good way to tell if my bed, and its glass plate, have stabilized and no more "heat soak" patience is required. Useful for preheating.

Thanks!!

### Solves the following problem:

Knowing when a bed (or hot end) has reached full heat-soaked stable operation.

meteyou, 2022-06-10:
just click on the heater and enable PWM in chart.

closed 2022-06-10 by meteyou

numindast, 2022-06-11:
I feel dumb. Thank you and sorry for the useless FR.