mainsail-crew/mainsail #897: Power On button when printer is disconnected.

LazaroFilm, 2022-06-12:
### Requested feature:

When the printer is off, I need to go into the top right power menu to turn it on, I would love a Power on button in the main view when the printer the Printer switch is detected in the OFF position.
[image]

### Solves the following problem:

Adding a Power On Printer button in the main dashboard area when the Printer switch is Off.

meteyou, 2022-06-12:
Something like this is on our to-do list, but I think I will only display one panel with "print is off -> switch on" or something similar.

LazaroFilm, 2022-06-13:
Amazing news!  .... waits... (jk. You guys are doing an amazing job already, no rush, take your time, I'll just be here, waiting patiently)

LazaroFilm, 2022-06-19:
> > ### Requested feature:
> 
> > When the printer is off, I need to go into the top right power menu to turn it on, I would love a Power on button in the main view when the printer the Printer switch is detected in the OFF position. [image]
> 
> > 
> 
> > ### Solves the following problem:
> 
> > Adding a Power On Printer button in the main dashboard area when the Printer switch is Off.
> 
> > 
> 
> > ### Additional information:
> 
> > _No response_
> 
> 
> 
> Hi, how to make Power on button ?

The image is a "photoshop" of what it could look like. To my understanding, as of now, it is not possible to have a power on button.

jordandrako, 2022-09-10:
Why wouldn't it be possible? Wouldn't it just be a matter of having an option to expose specific power devices on the dashboard (or even the top bar would be nice)

PR #1254 (2023-02-05): feat: add power button on dashboard to switch printer on
fixes #897

screenshot of the dashboard panel:
[image]

screenshot from the interface setting:
[image]
closed 2023-02-05 by meteyou