mainsail-crew/mainsail #904: Infinite RAM consumption over time on Firefox

pahofmann, 2022-06-15:
### Browser:

Firefox

### Device:

Desktop PC

### Operating System:

Linux

### What happened?

If mainsail has a webcam configured and mainsail is left open in firefox it keeps consuming more and more RAM till the system runs out of memory/swap.

This doesn't happen on chrome.

### What did you expect to happen instead?

Normal RAM usage.

### How to reproduce this bug?

Open Mainsail in Firefox on a printer with a webcam configures. 
Leave it running for a few hours.

th33xitus, 2022-06-15:
Which Mainsail version do you use?
With the release of 2.2.0 we fixed a few webcam related issues. Including a memory leak.

Joel-De, 2024-05-12:
This seems to still be an issue on 2.10 (updated to 2.11 last night), though it seems to be an edge case since it works fine most of the time. That said on 2 separate instances mainsail (on firefox) has consumed 7GB of ram and crashed the system. I am also using Crowsnest and have timelapses set up so there is that commonality. Don't have logs at this time but I'll create a new issue if it happens again with more details.

KungfuPancake, 2024-05-12:
I have the same issue on two separate computers, both Firefox, one on Archlinux and one on Windows 10. The problem is very likely coupled to the Webcam, as the problem is exacerbated by opening the large "Webcam" view.

Joel-De, 2024-05-12:
> I have the same issue on two separate computers, both Firefox, one on Archlinux and one on Windows 10. The problem is very likely coupled to the Webcam, as the problem is exacerbated by opening the large "Webcam" view.

@KungfuPancake I'm out of the city and won't have access to logs for about a week, if you're able to upload them and create an issue feel free to do that and link it here. I tried reproducing the issue today but wasn't able to get FireFox to use 100% of my RAM though it's entirely possible that I didn't wait long enough as I only had an hour.

meteyou, 2024-05-13:
@KungfuPancake @Joel-De which webcam modes do you use?

Joel-De, 2024-05-13:
@meteyou uStreamer & hyperlapse with the interval set to ~5s. I'm using 720p as the camera resolution so nothing too high.

meteyou, 2024-05-13:
@Joel-De which mode do you use in mainsail? The streamer itself doesn't matter...

Joel-De, 2024-05-13:
@meteyou I'm using local mode to host mainsail, used KIAUH to install the package and the rest of Klipper. I'm only using a single printer in my configuration.

meteyou, 2024-05-13:
@Joel-De i mean your webcam settings in mainsail itself. There are multiple types you could run your webcam in Mainsail.

meteyou, 2024-05-13:
@Joel-De no... i dont need your timelapse settings. timelapse will only be executing on the host (pi) itself. i need your webcam setting in mainsail! (no timelapse, no crowsnest...)

Joel-De, 2024-05-13:
@meteyou Are you referring to the items [here](https://docs.mainsail.xyz/overview/settings/webcams )? If not could you give me a link to the settings you're talking about? Thanks

meteyou, 2024-05-14:
@Joel-De yes, this page contains all webcam settings in Mainsail. Pls post a screenshot from each webcam.

Joel-De, 2024-05-17:
Here's a screenshot, I only have a single webcam setup on my system.

[image]

meteyou, 2024-05-17:
ok. can you double check, if you also get this error without a webcam in mainsail? pls don't forget to refresh 1 time after removing the webcam in this settings.

Joel-De, 2024-05-18:
I haven't been able to reproduce this issue with the camera disabled. That said I'm also not able to reliably reproduce it with it on either, although I did have this issue pop up yesterday after what I would estimate to be ~6 hours of runtime.

It doesn't seem to be slow building either - everything will be fine for several hours and it will abruptly balloon. I'll monitor it closely today and see if I can provide more hints. Are there any logs that would be useful for you to debug this?

meteyou, 2024-05-18:
Not really... This is the problem with this issue...

Joel-De, 2024-05-19:
Alright, tried it out on my desktop since I'm able to interact with the laptop that normally runs mainsail directly. Reproduced the bug though again it seems to appear randomly.

Memory snapshots of Mainsail didn't show any growth (~90MB of usage measured via Inspect -> Memory on firefox), in the meantime the main firefox process was growing pretty rapidly (~1GB -> ~2GB). From what firefox provides, "heap-unclassified" and "js-non-window" seem to be the areas in the main process consuming the most ram.

After about 7 hours my laptop that runs klipper and mainsail OOM'd and crashed. Didn't interact with it after bootup and it was on a screensaver after the first 30 minutes.

meteyou, 2024-05-19:
Normally, the browser go into sleep mode and mainsail should be on "pause"

Joel-De, 2024-05-19:
Yeah that's why I was surprised when it OOM'd while the screen had been off for several hours, I didn't think that was possible. I'll be switching from firefox to a chromium browser / kiosk viewer since some of these crashes have resulted in print failures, so I probably won't have too much more information to share here.

Feel free to re-open this issue or to create a new one if you feel that's justified.

Thanks

TECH-GEEK, 2024-08-27:
This bug is present in v2.12.0. I have one tab open with mainsail and it goes up to 16gb of ram usage for one tab in firefox in less than an hour.

meteyou, 2024-08-27:
@TECH-GEEK on how many places do you want to "spam" this now?

TECH-GEEK, 2024-08-27:
Sorry. I have autism and poor understanding of social skills.  I didn't mean to spam. Feel free to delete things

meteyou, 2024-08-27:
@TECH-GEEK ah ok. Np. Plus open the post in the discord support forum. I hope we can reproduce this issue, that we can fix it.

PR #1987 (2024-09-07): fix(webcam): fix memory leak in MJPEGStreamer client
## Description

This PR fixes a memory leak in the Mjpeg-Streamer client with Firefox.

## Related Tickets & Documents

fixes: #904 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none