mainsail-crew/mainsail #905: Graph the average PWM heater values

onde2rock, 2022-06-15:
### Requested feature:

I just discovered that we can graph the PWM values for the heater.
However, for me the PWM is all over the place and not very user-friendly :

[image]

Is is possible to instead plot the average PWM, like when hovering over the value:
[image]

I'm guessing this value is a rolling average over a minute ?

Since this value is already here, It should not be very difficult to add.
I'm trying to look in  ``src/components/charts/TempChart.vue`` and ``src/store/printer/tempHistory/actions.ts``, but I never used vue.js, so I'm a little lost.

Do you have any pointer where to start ?

meteyou, 2022-06-15:
to show PWM is to see issues with the PWM settings. for example, your chart looks like really really bad PWM values. if you have the right values, you should have close to a stable line.

so... it would be bad to make an average PWM value to display in the chart.

closed 2022-06-15 by meteyou

onde2rock, 2022-06-15:
Ok, makes sense.
That would explain why the auto pid tuning was not working as good as expected.
There must some problem in my setup, I will investigate.
Thanks

meteyou, 2022-06-15:
feel free to join our discord for helping!