mainsail-crew/mainsail #908: Option to hide Files & History Tabs in Status on Dashboard

rufinus, 2022-06-15:
### Requested feature:

Please add an option to hide "Files" and "History Tabs in Status on Dashboard. Every time you click on the Dashboard you land in Files... i dont care about files when a print is running. 

### Solves the following problem:

See the status of the print while printing.

th33xitus, 2022-06-16:
With https://github.com/mainsail-crew/mainsail/pull/907 we merged a change, so that every time you visit the Dashboard during a print, the Status tab will now be the active tab. That should fit your request, right?

meteyou, 2022-06-16:
If you hide these both sections and you have not loaded a file, you would only see the headline with nothing in the panel. It would break this panel complete.

rufinus, 2022-06-16:
@th33xitus thank you. sorry. i searched all issues but not pull request. yes this takes care of it.