mainsail-crew/mainsail #909: Text boxes not accepting values

bob92553, 2022-06-16:
### Browser:

Firefox

### Device:

Mobile device (Phone, Tablet or similar)

### Operating System:

macOS

### What happened?

While using my phone to start a print I typically put a temp in the provided text boxes in the main interface page. 
And as I click on the next temp box it starts warming up and accepts the value but now I have to type in the temp in the box. Click the enter key on the keyboard and about 1/3 of the time it accepts the value. Then I have to select the 2nd text box for temp enter the value and click the enter key on my phone keyboard before it will accept the desired value 

### What did you expect to happen instead?

I expected the value to stay as I click off as I always Do. To start warming up 

### How to reproduce this bug?

Open the main interface and put a temp value in the hotend then without hitting the enter key on your keyboard click the next temp value

bob92553, 2022-06-16:
This also happens on desktop I type in a value and as I hit tab to go to the next text box and it clears the value I put in and goes to 0

th33xitus, 2022-06-16:
Which Mainsail version do you use?

bob92553, 2022-06-16:
Will have to look when I get home. But I upgraded the the latest version.
When it just came out of beta

F0rt1s, 2022-06-16:
Same problem in Edge on Desktop
[image]

th33xitus, 2022-06-16:
> Same problem in Edge on Desktop

Again, this is no problem.
With the rework of that panel and the inputs we require the user to confirm the input with ENTER

bob92553, 2022-06-17:
This adds more work for my mobile experience. Can say I liked it better before. But can understand why the change was make will close the ticket

closed 2022-06-17 by bob92553