mainsail-crew/mainsail #920: continuous printing, expanding the functionality of job queue

sheldo26, 2022-06-19:
### Requested feature:

continuous printing, expanding the functionality of job queue,

add jobs to the continuous printing queue, multiply the number of times the job is printed consecutively
run cleaning script/macro after each print

th33xitus, 2022-06-19:
What is your exact feature request? Mainsail already has a job queue and continuous printing is a Moonraker feature: <https://moonraker.readthedocs.io/en/latest/configuration/#job_queue>
Running a macro after each print is also configures via the moonraker.conf.

Can you elaborate on the exact functionality you want to see improved?

sheldo26, 2022-06-19:
thank you, big help,

functionality to improve, when job is added to the queue, we are able to set number of times it reprints, before it moves to the next job in the list

example Job Queue:
Job A print: "10" times
Job B print: "3" times

this would reduce the over number of jobs in the queue, allowing users to better refine total prints when needing to continuous printing of the same model.

th33xitus, 2022-06-19:
I see, now i understand your idea. I will re-open this ticket. If we close it, we can't track that feature request anymore.

PR #1253 (2023-02-11): feat: add jobs to queue in batches
fixes #920
closed 2023-02-11 by meteyou