mainsail-crew/mainsail #928: Upgrade Klipper for multiple controllers

rlaferla, 2022-06-22:
### Requested feature:

Upgrade Klipper for multiple controllers.  Printers can have multiple controllers.  For example, I have three for my printer: Fysetc Spider, Huvud CAN and Raspberry Pi host.  The UI to upgrade the firmware only updates/upgrades one controller, the host.

th33xitus, 2022-06-22:
The host is not to be seen as an MCU like the other two you mention. The update button only allows you to update the contents of the Klipper folder on the host. To update controllers, you need to recompile the firmware and then flash it. Mainsail is just a web interface that is not capable of executing commands on the host. We need Klipper and Moonraker support for this sort of thing first, so that one day we can allow flashing of controllers via the Mainsail web interface.

meteyou, 2024-12-30:
I will close this PR, because it need first an implemention in Moonraker (and maybe something in Klipper) first. Feel free to post any progress here to reopen this FR.

closed 2024-12-30 by meteyou