mainsail-crew/mainsail #929: Power toggle initially grayed out on secondary printer instance

numanair, 2022-06-22:
### Mainsail Version:

v2.1.2

### Browser:

Firefox

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

When trying to switch to and then power on my secondary printer, the power toggle is grayed out. If the power state is switched externally with the 2nd printer active in Mainsail, the toggle is no longer grayed out. I am 80% sure this used to work.

### What did you expect to happen instead?

Can toggle printer power.

### How to reproduce this bug?

1. Load Mainsail on 1st printer
2. Select 2nd printer from dropdown in Mainsail 
3. Attempt to toggle power from the Mainsail power menu

### Additional information:

1st printer serves Mainsail while the 2nd printer is just Moonraker and Klipper. If this is not the ideal setup I am open to suggestions.

meteyou, 2022-06-22:
pls upload mainsail and test it again

numanair, 2022-06-26:
I am actually still having the same issue after all.

numanair, 2022-07-15:
I have narrowed down the issue slightly. The button is only grayed out after starting a print on the first printer. I do have `locked_while_printing` in my Moonraker config.

meteyou, 2022-07-15:
pls post your moonraker configs of both printers. do you also updated to v2.2.x?

numanair, 2022-07-15:
Yeah, I am on v2.2.1.
[moonraker(1).txt](https://github.com/mainsail-crew/mainsail/files/9124496/moonraker.1.txt)
[moonraker(2).txt](https://github.com/mainsail-crew/mainsail/files/9124495/moonraker.2.txt)

meteyou, 2022-07-15:
both are the same moonraker config file. port 7125

numanair, 2022-07-15:
Should I change the port? They are from two different moonraker instances each running on their own pi. Two printers, two pi's. I switched away from multi instance/single pi as I was having issues with crashes.

meteyou, 2022-07-15:
ah ok. sry. mixed it up. thought you have both on the same pi. then its all ok

meteyou, 2022-07-17:
pls upload also both moonraker.log files. i tested it with your config, but it works for me.

numanair, 2022-07-18:
moonraker.logs:
[moonraker(1).log](https://github.com/mainsail-crew/mainsail/files/9128847/moonraker.1.log)
[moonraker(2).log](https://github.com/mainsail-crew/mainsail/files/9128846/moonraker.2.log)

Try these repro steps:
(ender3=printer1 running mainsail, MPSM=printer2)

1. Connect to mainsail (pi1)
2. Switch to printer2 from the top dropdown 
[image]

3. Power on printer2 from the power menu 
[image]

4. Start a print
5. At this point printer 1 in the top dropdown shows as disconnected & grayed out. Not sure if this is related. 
[image]

6. Switch to printer 1 from the printers menu by first clicking "reconnect" and then "switch to printer" 
[image] 

[image]

7. Power device cannot be toggled from the menu. 
[image]

8. Reload the page. Power can now be toggled. 
[image]

9. Switching between printers with the "switch to printer" button again grays out the toggle.

meteyou, 2022-07-18:
I think it is more an issue, that Moonraker cannot connect to the plug ant that's why it is disabled.

numanair, 2022-07-28:
I am not so sure about that. If that was the case, why does the printing state affect the other printer's toggle?

numanair, 2022-09-14:
Fixed with v2.3

closed 2022-09-14 by numanair

meteyou, 2022-09-14:
thx for your feedback!