mainsail-crew/mainsail #930: [FR] Exclude object GUI support

watzon, 2022-06-23:
### Requested feature:

Recently merged into klipper was [support ](https://github.com/Klipper3d/klipper/pull/5413) for excluding objects from a print. The `exclude_objects` config property gives you access to some special GCODE that will cancel certain objects from printing, while allowing the rest of the print to continue as normal.

It would be nice if the Mainsail UI supported this in some way. Personally I think the GCODE Preview window would be the perfect place for this if possible. I could see this being implemented in one of two ways.

1. Probably the hardest option would be to right click the item in the GCODE preview window, and then choose to ignore the item via context menu.
2. Have a list of items being printed off to the side. On hover highlight the item in the preview window and show a button for canceling.

### Solves the following problem:

The biggest one I can think of is that sometimes when printing multiple items at a time you'll have one fail, forcing you to stop the print before that failure causes the other pieces to fail. Being able to just cancel the one item that failed and save the rest of the print sounds amazing.

Octoprint has had this for a while by use of a plugin.

### Additional information:

Related is #169 which was ahead of its time

meteyou, 2022-06-23:
all what you wrote is already included in mainsail since v2.1. only the preview thumbnail is not "clickable", because it is only a picture. the only thing, what i have on my todo list, is to integrate it also in the 3d gcode viewer. it only display each object atm, but you cannot cancel it in the viewer. only on the dashboard.

pls read the docs and test it. https://docs.mainsail.xyz/features/exclude_objects

closed 2022-06-23 by meteyou

th33xitus, 2022-06-23:
I think that is exactly what he means. Expand the exclude object feature and allow users to cancel via the gcode viewer.

So we could reopen this issue, reword the title and use it for tracking?

watzon, 2022-06-24:
Yeah when I first wrote this FR I didn't know that exclude object support had been added recently, but I think it would be good to keep around for the gcode viewer integration.